vdeconinck/QC3Control #1: Is the DM low pin required for QC3 mode?

CameronLamont, 2017-10-08:
I have been attempting to use the QC3 library - I may be misunderstanding the code but it appears to require the 3rd DMlow pin within the code? For example

if (_DmLoPin == 0) return QC_ERROR_QC2_ONLY;

QCControl.cpp lines 104, 127, 168

There are other references to this value including digitalWrites to it (the 3 dm...V() functions).

With this line in-place it never gets to the QC3 parts of the code. If I use the 2 arg constructor it sets it to zero (Line 19). I haven't yet attempted a 3 pin setup. (I was wondering if this was an older version of the code compared to your video and descriptions?).

Thanks for the work you've done on this I'm keen to fire it up! I have a QC3 source that apparently goes to 20V so I can test if that helps (I remember you mentioning that you didn't have hardware) 

Sorry for posting in the wrong place, I can't do anything more as it is locked

vdeconinck, 2017-10-08:
You're right. Somehow an old test version got commited. 
I'm updating and checking things right now and will report back here.

vdeconinck, 2017-10-08:
OK, I just released v1.3.0 which should fix the issue. According to the [library manager FAQ](https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ), it should be proposed in the Arduino IDE within one hour.
And to answer your question, no, the 3rd pin is not required with the latest version :-). Sorry for messing up the previous version :-(

I suggest that once you confirm it to be working (up to 12V) with v1.3.0, you close this issue and open a separate one for the 20V feature (but I'm already having a look right now :-)).

CameronLamont, 2017-10-08:
Thankyou, note the examples also have the ‘3 pin’ constructor for the ‘basic recommended’ – your github site has a newer sample

vdeconinck, 2017-10-08:
Yes indeed. v1.3.0 should reflect what you now see on github, examples included.

CameronLamont, 2017-10-08:
Thanks will get into it now

vdeconinck, 2017-10-08:
I opened #2 for the 20V "class B" mode. 
Please just report here about the basic "class A" behaviour and close it if it works.

CameronLamont, 2017-10-09:
thanks will give it a try this evening.

I have a wall adapter and a powerbank to try with - the latter is 20V
compatible (apparently)

CameronLamont, 2017-10-09:
I now have the code setup and it appears to be working! I had debug statements in my checkout repo but I blew them away by mistake! (had considered sending a pull request).

closed 2017-10-09 by CameronLamont