vdeconinck/QC3Control #2: Support Quick Charge Class B (up to 20V)

vdeconinck, 2017-10-08:
As requested in issue #1, work has been done to support Quick Charge "Class B" chargers, to allow voltage outputs up to 20V.
The changes with that feature are now commited. All that is required to enable Class B mode is to call "begin(true);" at startup.

To test these changes, first delete the QC3Control fodler from your sketchbook's "libraries" folder, then download the zip using QC3Control's homepage button "Clone or download", and import it in the IDE using "Sketch > Include library >Add .ZIP library..."

Please report the results here...

CameronLamont, 2017-10-09:
I have tried with the new example code but could not go past 12V.  

After rechecking my power pack cannot exceed 12V using QC3.0 but can go to 20V using USB Power Delivery mode.  When I ran it the voltages did step up as per QC3.0 but were ignored hire than 12V it seems.

So this is not necessarily a failure for the library.

vdeconinck, 2017-10-09:
Hmmm. Can you confirm you used the classBControl example or you called begin(true); at the beginning of your sketch ? 

To decide for sure, could I ask you to do the following test:
1) flash the following minimal sketch and verify the output is 12V
```
void setup() {
  begin(true);
  set12V();
}
void loop() {}
```
2) If so, replace set12V(); by set20V(); and check the output voltage.
If it's not 20V, please measure the voltage on D+ and D-. 
If they are somewhere around 3.3V as they should, I'm afraid the powerbank is not QC3 class B compliant...

CameronLamont, 2017-10-09:
>    1. If so, replace set12V(); by set20V(); and check the output voltage.

I tried a modified version of the basicRecommended with the extra argument
as well as the classB example.

I definitely saw the voltage - using a multimeter not just the value in the
qc class - step gradually downwards by 200mV.

I will try it again as you suggest and let you know. Perhaps I wasn't
giving it enough time to step up gradually - e.g. 13V

vdeconinck, 2017-10-10:
FWIW, I released v1.4.0 with the class B changes. 
Hopefully someone with a QC3 class B charger will confirm it works...

closed 2019-02-03 by vdeconinck