vdeconinck/QC3Control #3: Arduino unable to compile/execute 'inline' functions getVoltage() , getMilliVoltage()

CameronLamont, 2017-10-09:
Not sure why but I am unable to compile code if I try and call these functions in QC3Control.cpp (See below).  They are defined without 'inline' in the .h signature and are public

 QC3Control.cpp v.1.3.0
L234
inline double QC3Control::getVoltage(){
L238
inline unsigned int QC3Control::getMilliVoltage(){

When I hit verify or upload:
e.g. undefined reference to `QC3Control::getMilliVoltage()'

My workaround was to delete 'inline' from the cpp file and it compiles and runs. I'm using Arduino IDE 1.8.1 on Win10 x64.  From what I gather 'inline' is a bit like a #define style macro? (pre-processed?) - any specific reason to use inline? (I didn't know it existed until reading about it for this).  I thought perhaps you are using a different compiler / OS.

When I googled around I found a few results like this: https://forum.arduino.cc/index.php?topic=330924.0  - but this is from ~2015.

vdeconinck, 2017-10-09:
Hmmm, yeah. I probably didn't test these functions recently. I admit I kept them to be backwards compatible with QC2Control but I think they are misleading because they don't report the actual output voltage, they just report the last voltage that was requested.
Now that's not an excuse and I will fix that of course (probably tomorrow).
In the meantime, you can drop the "inline" prefix of course. 
All it does is instruct the compiler to not make that code a traditional function called from many places, but just to "copy/paste" the implementation each time the function is called. You gain a few cycles at each call (because you don't spend time "saving the context" before the call and restoring it upon return) at the expense of a slightly larger code (due to the multiple copies).
But for such a simple project, optimization does not make sense, so I'll probably drop the "inline"s altogether.
Thanks for reporting the issue.

CameronLamont, 2017-10-09:
no worries - this was actually what I encountered first but hadn't
attempted to raise at the time.

I was wondering about how the arduino actually knows for sure the power
source is 'shaking hands' - e.g. does it reply somehow

vdeconinck, 2017-10-09:
Nope, the Arduino doesn't "know" anything, it just blindly fiddles with voltages, hoping that there is a QC source listening at the other end.
One possibility I thought of was to monitor the output voltage. Not only would it confirm that the charger is behaving as expected, it would also allow to implement [remote sense](http://power-topics.blogspot.be/2008/07/what-is-remote-sensing.html) and increase voltage to compensate for the loss in the lines.
But it would require more pins and I thought it was out of the scope of the project. Of course, you can always implement voltage monitoring as part of the sketch that uses the library...

CameronLamont, 2017-10-10:
Yes I had similar thoughts - either/both 'listening' if the QC source
acknowledges anything and potentially an external sensor too. I like that
it uses 2 pins - I want to try with an esp8266 (I have the one with 2
gpio). The other protocol I have looked at is USB power delivery; both
extensions would suit if the protocols are compatible - ext voltage avoids
this but more pins (also not dure about larger tha arduino voltages)

I will give the two pin handshake method a go but am still learning on the
electronics side :) (e.g. Not sure yet how to detect the short). USB seems
orders of magnitude more complex but I think the protocol is more open.

My 3 use cases - portable rpi3 (maybe have the pi talk to qc to up it
slightly to 5.2V), portable kinect (12V) and part of a lithium battery
charger (in this case 13.6V). Having a (hopefully tiny) in line microusb
adapter would be cool - especially if it can speak pd as well.

vdeconinck, 2017-10-10:
In any case, there is no "dialog" with a request and an acknowledge. Basically, you just try and see the result, and that result can be monitored.
As for the ESP8266, keep in mind that it's a 3.3V device. Even if you have a wemos, wittycloud or other adapter that includes a voltage regulator from 5V, the GPIOs are still 3.3V, so maybe the resistor values will have to be adapted to compensate for the difference...

CameronLamont, 2017-10-10:
Ok good to know. I have a voltage regulator setup already. Hadn't yet
adjusted your spreadsheet yet :)

vdeconinck, 2017-10-10:
Fix for "inline" is now commited in release 1.4.0. 
Please don't hesistate to report your results here.

closed 2019-02-03 by vdeconinck