vdeconinck/QC3Control #8: Trinket 3.3v support

theodonkulus, 2018-09-25:
Hi first of I just wanted to say this is awesome work! Your work on this project and you've opened the door to plenty of devs to use this in their project without having to reinvent the wheel for a reusable, LiPo PSU in their projects without relying on hobby alternatives!

I was wondering, is it possible to use this with a trinket 3.3V? It seems like your data signals are going between the desired 0 - 3.3V with your resistor configuration from your blog post and your diagram. The trinket has two GPIO pins directly connected to the usb D+/D- pins that go between 0 and 3.3 V. Would this be enough to operate this library without the resistor block instead? Worst case can't we find a way to use the fact that the resistance changes on the 0.6v handshake to 20K between the pins, while using the other pins after the startup sequence is complete?

theodonkulus, 2018-09-25:
Actually disregard, my daily coffee has worn off. I think we can actually use the supply from the 3V output pin and two resistors to set a pull up of 0.6V between the two in some way when the signal is a logical low on those pins..hm. refering to this for the 3.3v layout. 

https://learn.adafruit.com/introducing-trinket/pinouts

vdeconinck, 2018-09-25:
Hi. No problem :-). 
Could you please close the issue if it's not relevant anymore.
BTW: Note that I'm not the first one who fiddled with resistors to control a QC charger. I just adapted https://github.com/septillion-git/QC2Control by Septillion to make it compatible with QuickCharge 3 and its awesome "incremental" modes :-)

theodonkulus, 2018-09-28:
Yes i know. I've seen the work of both of you.

closed 2018-09-28 by theodonkulus