vdeconinck/QC3Control #9: Can't change voltage with 2 wire configuration.

DanielArnett, 2019-02-01:
Just an interesting observation: when I use the 2 pin configuration I can change the voltage on my Samsung fastcharger, but not on my Anker battery. But once I switch back to the legacy system I can use the Anker battery. Any ideas why? It says it's a QC3 compatible battery.

Thanks to all who helped make this awesome library.

DanielArnett, 2019-02-02:
> The problem with quickcharge is that there is no public official
> documentation, so many implementations are based on the little
> documentations gathered from various datasheets. It does *usually* work,
> but occasionally the "detection tolerances" do not match up.
>
> Example:
> Imagine the Anker battery draws slightly more current from the data lines
> than the Samsung charger. Thus the voltage across the voltage devider would
> drop slightly, which may not be enough to be detected by the QuickCharge
> detection circuit.
>
> Possible Solution:
> You may try to reduce R3 slightly (9.1kΩ maybe?) or half the value of R3
> and R4 to 5.1 kΩ and 750Ω maybe.
>
> Disclaimer:
> I've done some research on QC3 a while ago, but never used this library or
> either of your devices. Although I firmly believe the modifications above
> won't damage your devices, I can *not* guarantee it, so proceed at your
> own risk.

Neat, thanks for the feedback!

vdeconinck, 2019-02-03:
Hi. As nqtronix says, QC specs are not public, so this project is based on leaked docs and testing. 
It could be indeed that the levels are wrong and fiddling with resistor values would work in 2-pin setup (don't go below 220 ohm though), but the issue could also be caused by the 2-pin setup not adhering perfectlty to the QC logic, because it doesn't leave the D- floating. See "_Note: Why keep 2 circuits and 2 implementations ?_" at the end of [my blog post](http://blog.deconinck.info/post/2017/08/09/Turning-a-Quick-Charge-3.0-charger-into-a-variable-voltage-power-supply).
To summarize, there's a reason why I didn't ditch the 3-pin implementation :-)
KR,
Vincent

closed 2019-02-03 by vdeconinck

DanielArnett, 2019-02-03:
Thanks @vdeconinck! Do you know the difference with PowerIQ from Anker? It would be great to have something that works with both QC and PowerIQ.
http://community.anker.com/t/anker-faq-10-things-you-need-to-know-about-anker/57477