vdeconinck/QC3Control #11: Need help!! Not working

im-AMS, 2020-07-23:
Hi,

I tried running the same rig as u have mentioned,
Im not able to figure out why Its not working for me

what I know

- Im sure its a quick charger, i have seen rating(5v,9v,12v) and also my phone charges with quick charge
- the voltage levels at D+ D- seem ok, i have also tried reversing D+ and D-, no luck

Can u please help me out!!

im-AMS, 2020-07-23:
I feel embarissed now, I tried the legacy version and it worked...Sorry tough!!

BTW thanks for this!!

closed 2020-07-23 by im-AMS

vdeconinck, 2020-07-23:
Yeah, these technologies are based on grey areas of the USB specifications and this is basically a hack. and a bit of trial and error is often needed.
Anyway, great to know it is sorted out :-).