vdeconinck/QC3Control #12: Charger acting weird

Matszwe02, 2020-10-26:
When I upload this code, firstly my charger has nice 5v, and then it switches to 2,5V instead of 9V. I tested this with 2 chargers, it always happen.

vdeconinck, 2020-10-26:
This project is a library, so I'm not sure what calling code you are using. Could you please post that code ?
Or do you refer to an example ?
Also, can you indicate which circuit you are using ?

Matszwe02, 2020-10-27:
I tested with QC 2.0 example sketch and every circuit that you posted

vdeconinck, 2020-10-27:
That's wierd... Can you try using just the following sketch:
```
void setup() {
  quickCharge.set9V();
}
void loop() {}
```
?
Also, what circuit do you power with your charger ?

Matszwe02, 2020-10-27:
This charger is connected only to a voltmeter. Does it matter if i don't have any load?

vdeconinck, 2020-10-27:
I guess not, but I was just wondering... 
Could you please test the minimal code above ?

Matszwe02, 2020-10-27:
I already disassembled this circuit sad that it didn't work, so reassembling may take a while. I'll text you in a while.

Matszwe02, 2020-10-27:
Weird. It doesn't work, but it worked yesterday. Maybe I messed with these resistors. Anyway, I didn't tell you that, when I connected 2 potentiometers directly to USB data lines and set 0.6 and 3.3V (i found a table with these values), this charger also switched to something near 2.5V. However, when I charge a smartphone, it normally switches to 9V. Oh yes, and this is 2.0 QC charger, but QC3.0 is backward compatible.

vicnevicne, 2020-10-27:
Hmm.
First things first, if it's a QC2 charger, and it outputs 2.5V no matter how, then something's very wrong. QC2 chargers are supposed to only output 5V, 9V, 12V or optionally 20V. 
2.5V being half of 5V, it could be that there is a short somewhere and you're measuring half the standard 5V through a dividing bridge or something. Or maybe the charger is defective.
Anyway, to rule out any issue with the QC3 lib, I'd be tempted to direct you to the original QC2Control lib at https://github.com/septillion-git/QC2Control . 
If that one works perfectly, I'd suggest you keep it, but don't hesitate to report here, along with the voltages measured on the D+ and D- wires.

vdeconinck, 2020-10-27:
Hmm.
First things first, if it's a QC2 charger, and it outputs 2.5V no matter how, then something's very wrong. QC2 chargers are supposed to only output 5V, 9V, 12V or optionally 20V.
2.5V being half of 5V, it could be that there is a short somewhere and you're measuring half the standard 5V through a dividing bridge or something. Or maybe the charger is defective.
Anyway, to rule out any issue with the QC3 lib, I'd be tempted to direct you to the original QC2Control lib at https://github.com/septillion-git/QC2Control .
If that one works perfectly, I'd suggest you keep it, but don't hesitate to report here, along with the voltages measured on the D+ and D- wires.

Matszwe02, 2020-11-03:
Hi again. I bought a QC 3.0 charger and it works perfectly! I still don't know how this QC2.0 charger acted in that way.

closed 2020-11-03 by Matszwe02

Matszwe02, 2020-11-03:
Thanks for your help!

vdeconinck, 2020-11-03:
Great !
Thanks for reporting back :-)