vdeconinck/QC3Control #13: Resistors for 3.3v boards

allenchak, 2020-11-04:
Tested the sketch for Pro mini 5V, it is work and great. 
But when I run on Pro mini 3.3v, it's just working with some QC2 power supply.

May I know the corresponding resistor values for Arduino Pro Mini 3.3v ?

vdeconinck, 2020-11-04:
Hi,
Interesting question :-)
**Disclaimer:** I have not tested anything of the following. 
I would try with those values:
R1 = R3 = 4700 Orm
R2 = R4 = 1500 Ohm
R5 = R6 = 100 Ohm
This is just the result of some fiddling with the values in the blue cells of the spreadsheet available in the blog post and adapted for variable values of Vcc, attached here: [Resistor calculator variable Vcc.zip](https://github.com/vdeconinck/QC3Control/files/5490865/Resistor.calculator.variable.Vcc.zip)
The goal is to bring the 5 big yellow dots close to green dotted lines, each in its own area (delimited by black lines) :-)
I guess you can try those values and see how it goes. It shouldn't break anything (although I take no responsibility as once again this was not tested).
Don't hesitate to report your results :-)

allenchak, 2020-11-12:
Yes, @vdeconinck, work with Pro Mini 3.3v and ATtiny13A. Thank you very much.

closed 2020-11-12 by allenchak

vicnevicne, 2020-11-12:
Great ! Thanks for confirming :-)

vdeconinck, 2020-11-12:
Great! Thanks for confirming.

allenchak, 2020-12-31:
[image]
All features are work thanks again.

vdeconinck, 2020-12-31:
Nice box ! Excellent ! :-)

Bleach665, 2023-04-21:
> Yes, @vdeconinck, work with Pro Mini 3.3v and ATtiny13A. Thank you very much.

@allenchak Did you take 3.3 or 5 V for VCC for this resistor values?
````
R1 = R3 = 4700 Orm
R2 = R4 = 1500 Ohm
R5 = R6 = 100 Ohm
````